embeddedartistry/arduino-printf #17: include error when using in platformio

ProjectPatatoe, 2020-08-23:
When using inside of PlatformIO I get this error about dependency.

```
cannot open source file "../extras/printf/printf.h" (dependency of "LibPrintf.h")C/C++(1696)
```
```
.pio\libdeps\nanoatmega328new\LibPrintf\src\LibPrintf.h:5:37: fatal error: ../extras/printf/printf.h: No such file or directory
```

phillipjohnston, 2020-08-23:
Hello,

I’ve never used PlatformIO, so I can’t provide a helpful answer at the moment. 

I’m curious: do those files actually exist wherever the Arduino-printf files are kept?

studionebula, 2020-09-13:
I don't know how PlatformIO interfaces to github, but the problem appears to be that when PlatformIO pulls the library into the project, it does not grab the extras folder. Thus the compilation fails because the header file can't be found, as the original poster described. Sorry that I can't tell you how to fix it; this kind of seems like PlatformIO's issue rather than yours.

EmperorArthur, 2020-11-24:
There is a work around!

Instead of relying on platformio's own (faulty) packages, just use the github archive directly.

Put this in your platformio.ini file:
```
lib_deps = 
	LibPrintf=https://github.com/embeddedartistry/arduino-printf/archive/1.1.4.tar.gz
```

uchnz, 2022-03-16:
> 

worked for me. Thanks!