embeddedartistry/arduino-printf #19: printf_init(Serial3) does not work

asenchristov, 2021-01-14:
(Using RobotDyn MEGA+WiFi R3 ATmega2560+ESP8266, flash 32MB, USB-TTL CH340G, Micro-USB)
the switches are set in a way that ATmega2560 talks to ESP8266 using **Sereal3**.
I tried the example (specify_print_class.ino), (of course changing Serial1 to Serial3 inn two places) but it did not work, it did not switch to using Serial3.

On the other hand, if I go and edit LibPrintf/src/LibPrintf.cpp , line 15, and hard-code Serial3 there all the text is correctly sent to Serial3.

AlexKlimaj, 2021-01-22:
I am seeing the same thing trying to use `printf_init(Serial1);` on the B-L072Z-LRWAN1 using 
https://github.com/GrumpyOldPizza/ArduinoCore-stm32l0

phillipjohnston, 2021-10-01:
Sorry I neglected this so long - got filed when I was snowed in with a newborn, and then I didn't have any reminders. Anyway, I confirmed this is happening to me as well - when I call `printf_init`, then the instance reference isn't updated. Converting everything to pointers internally works well. I don't love that, but it is my fallback plan. 

My best guess at this moment is that the compiler is optimizing the reference away, even though we have the possibility of updating it. Will try to confirm that and see what exactly is at play.

phillipjohnston, 2021-10-01:
Nope, turns out I just didn't understand that you can't retarget C++ references!

PR #23 (2021-10-01): Update internal Print variable to a pointer
I didn't realize you couldn't retarget a reference variable after it was initialized, so now Print is a pointer.

Resolves #19
commit 09e5ed28 (2021-10-01): Update internal Print variable to a pointer

I didn't realize you couldn't retarget a reference variable after it was initialized, so now Print is a pointer.

Resolves #19
closed 2021-10-01 by phillipjohnston