embeddedartistry/arduino-printf #20: error: '::printf', '::sprintf', '::vprintf' has not been declared for ESP8266

Longoon12000, 2021-03-18:
Hello

I am using an apparently Arduino compatible ESP8266 module (https://github.com/esp8266/Arduino). When building with this library I get the following error message:

```
In file included from c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\random:39:0,
                 from c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algo.h:65,
                 from c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\algorithm:62,
                 from C:\Users\user\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/Arduino.h:238,
                 from C:\Users\user\Documents\Arduino\libraries\LibPrintf\src\LibPrintf.cpp:2:
c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\cstdio:121:11: error: '::printf' has not been declared
   using ::printf;
           ^
c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\cstdio:131:11: error: '::sprintf' has not been declared
   using ::sprintf;
           ^
c:\users\user\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506\xtensa-lx106-elf\include\c++\4.8.2\cstdio:137:11: error: '::vprintf' has not been declared
   using ::vprintf;
           ^
exit status 1
Error compiling for board Generic ESP8266 Module.
```

It does work when building for Arduino modules.
Is there any way to get this library running on ESP8266?

Thank you very much!

phillipjohnston, 2021-03-19:
I will need to purchase an ESP8266 and explore this further. I've had a lot of problems with this library and ESP in general. Also - doesn't the ESP core already provide a printf? I think that was a problem we ran into with another user.

Longoon12000, 2021-03-19:
Oh wow I did not realize that ESP8266 already contains printf which prints to Serial by default.
If the ESP documentation wasn't absolute ass I would probably eventually have found it. :|
Thanks for the hint.

closed 2021-03-19 by Longoon12000