embeddedartistry/arduino-printf #21: PlatformIO doesn't like the reference to files in the "extras" folder

RalphBacon, 2021-09-05:
PlatformIO can include this library in the _plaformio.ini_ file like this:

lib_deps = embeddedartistry/LibPrintf@^1.1.3

But compilations fails as there are references to an "_extras_" library that is not part of (or subordinate to) the "_src_" folder and is thus considered outside of the source files and not downloaded.

Please consider putting the extras files (config stuff, it looks like) in the _src_ folder so PlatformIO can find it.

Workaround: copy the entire GitHub repository into the "lib" folder; this however, freezes the library and will not get updates.

phillipjohnston, 2021-09-05:
Then it doesn't work with its intended environment, Arduino. A workaround for Platformio's packaging problem can be found here: https://github.com/embeddedartistry/arduino-printf/issues/17#issuecomment-733244442

phillipjohnston, 2021-09-05:
Closing as a duplicate.

closed 2021-09-05 by phillipjohnston