embeddedartistry/arduino-printf #22: Not compatible with Teensy!

jcernato, 2021-09-30:
This does not work with PIO and Teensy 3.2.

At first I thought my teensy got bricked (not responding), even my host system (ubuntu) crashed.
Turns out that teensy uses the HID protocol and the lib somehow messes with that, therefore all HID devices on my ubuntu started behaving weirdly.
After rebooting the system, holding down the "programming button" on the teensy before an while plugging it in, I could re-flash the microcontroller which now works perfectly fine again.

Somehow - however - this event messed with a USB-hub. I have a Dell-da300 port replicator (which still works fine btw) and if connected, the kernel now gives me an error message every five seconds:
"[ ... ] usb usb4-port1: Cannot enable. Maybe the USB cable is bad?"

Not sure how this is related, but it didn't happen before...

phillipjohnston, 2021-09-30:
Did this happen after a recent update, or upon the first time trying the library? Do you have any source code I can take a look at?

I have a hard time seeing how this library could do such things, this mostly wraps around the built in Serial class by formatting the printf string information and calling Serial.print(). But stranger things have happened. 

I'll give this a shot with some of my hardware, though I don't think I have a 3.2 on hand.

jcernato, 2021-09-30:
After playing around with your provided examples and some minor tweaks within my project (probably the most important was to include <LibPrintf.h>) it does indeed work now!
My apologies, I was really looking forward to this function.

https://www.pjrc.com/teensy/troubleshoot.html
states that teensy uses the HID protocol to handle the usb connection, interesting that a software bug in the teensy could break the HID handling on a linux-host (mice and keyboards ceased working properly).
Still no clue what happened to my usb-port-replicator, the kernel-message is quite annoying...

closed 2021-09-30 by jcernato