embeddedartistry/arduino-printf #30: printf does not output anything with the %c format charater

akoerner, 2022-03-15:
## Problem
printf("%c", 'A');  outputs nothing

## Environment 
Arduino UNO
Windows 10 
Arduino IDE 1.18.13
LibPrintf 1.1.3

## Expected Behavior

Given the following program compiled and uploaded to the Arduino I would expect the serial console to show a single "A" character

```
#include <LibPrintf.h>\\LibPrintf 1.1.3

void setup() {
  Serial.begin(9600);
  printf("%c", "A");
}

void loop() {
  
}
```

## Actual Behavior 

nothing outputs to the serial console.
 Installing **LibPrintf Version 1.1.2** results in the correct and expected behavior.

phillipjohnston, 2022-03-15:
Huh, versioning must be messed up somewhere, as I'm expecting versions in the 1.2.x range. I'll look into this - I expected the problem to be fixed with this release. https://github.com/embeddedartistry/arduino-printf/releases/tag/1.2.8

What's happening is that GCC is optimizing that printf call to putchar, which is either not implemented or not implemented correctly.

akoerner, 2022-03-16:
> 1.2.6 (latest available in Library Manager) has the issue, I am spinning 1.2.10 with an updated library.properties file which will have the fix.

1.2.10 is now available via the Library Manager and resolves the issue.

closed 2022-06-07 by phillipjohnston