embeddedartistry/arduino-printf #33: what is the macro to disable all prints

joseph-vedadi, 2022-05-19:
Is there any macro where I could diable all prints, and basically fully remove the lib, when I am not debugging

phillipjohnston, 2022-05-20:
Not at present. What I have done in the past is to make an internal print macro, such as `debug_print`. If a definition is present, such as `DISABLE_DEBUG_PRINTS`, it is defined to an empty value, otherwise it simply aliases `printf`.

I'll work on implementing something like that in the Arduino library wrapper next week.

phillipjohnston, 2022-05-20:
Alternatively, our logging library already supports this behavior (as well as configurable log levels): https://github.com/embeddedartistry/arduino-logger

joseph-vedadi, 2022-05-20:
> Not at present. What I have done in the past is to make an internal print macro, such as `debug_print`. If a definition is present, such as `DISABLE_DEBUG_PRINTS`, it is defined to an empty value, [...]

Thank you..: 
I am not such an expert on macros
How does your macro looks like to be able to print multiple values

phillipjohnston, 2022-05-20:
General form:

```
#ifdef DISABLE_DEBUG_PRINT
#define debug_print(...)
#else
#define debug_print(...) printf(...)
#endif
```

phillipjohnston, 2022-05-20:
Whoops, corrected `printf(...)` to `printf(__VA_ARGS__)`

phillipjohnston, 2022-05-22:
> thank you so much .. Do I need to put that in all files ... ?
> I put this on my mian, and other files do not see it ..

Sometimes i do this on a per-file basis for fine grained control. For application wide control i would make a new header like debug_print.h. In that you could include the printf header. Your source files will include the debug_printf.h header

Cheers,

Phillip

PR #35 (2022-06-07): Add ability to remove all printf-related functions from a program.
Fixes #33
closed 2022-06-07 by phillipjohnston