embeddedartistry/arduino-printf #41: Issues building in platformio for ESP32.

Destarianon, 2023-03-07:
I am running into an issue building an ESP32 project in platformio with some error messages that I have never seen before.

I have attempted to rollback the Espressif32 platform version from 6.0.1 to various other versions up to 2 years in the past, which also would have downgraded the toolchain versions, but with no improvement.

Is this something anyone has seen before?

Error output below:
```
In file included from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\ext\string_conversions.h:43,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6400,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\string:52,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:/Users/Destari/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:49,
                 from C:/Users/Destari/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:174,
                 from .pio/libdeps/adafruit_feather_esp32_v2/LibPrintf/src/LibPrintf.cpp:4:
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:127:11: error: '::printf' has not been declared
   using ::printf;
           ^~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:137:11: error: '::sprintf' has not been declared
   using ::sprintf;
           ^~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:145:11: error: '::vprintf' has not been declared
   using ::vprintf;
           ^~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:146:11: error: '::vsprintf' has not been declared
   using ::vsprintf;
           ^~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:175:11: error: '::snprintf' has not been declared
   using ::snprintf;
           ^~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:178:11: error: '::vsnprintf' has not been declared
   using ::vsnprintf;
           ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:185:22: error: '__gnu_cxx::snprintf' has not been declared
   using ::__gnu_cxx::snprintf;
                      ^~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\cstdio:188:22: error: '__gnu_cxx::vsnprintf' has not been declared
   using ::__gnu_cxx::vsnprintf;
                      ^~~~~~~~~
In file included from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\string:52,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\stdexcept:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\array:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\tuple:39,
                 from c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\functional:54,
                 from C:/Users/Destari/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.h:49,
                 from C:/Users/Destari/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:174,
                 from .pio/libdeps/adafruit_feather_esp32_v2/LibPrintf/src/LibPrintf.cpp:4:
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(int)':
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6454:50: error: 'vsnprintf' is not a member of 'std'
   { return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, 4 * sizeof(int),
                                                  ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6454:50: note: suggested alternative: 'isprint'
   { return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, 4 * sizeof(int),
                                                  ^~~~~~~~~
                                                  isprint
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(unsigned int)':
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6459:50: error: 'vsnprintf' is not a member of 'std'
   { return __gnu_cxx::__to_xstring<string>(&std::vsnprintf,
                                                  ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6459:50: note: suggested alternative: 'isprint'
   { return __gnu_cxx::__to_xstring<string>(&std::vsnprintf,
                                                  ^~~~~~~~~
                                                  isprint
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(long int)':
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6465:50: error: 'vsnprintf' is not a member of 'std'
   { return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, 4 * sizeof(long),
                                                  ^~~~~~~~~
[... 30 lines cut ...]
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6491:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(double)':
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6500:50: error: 'vsnprintf' is not a member of 'std'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6500:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h: In function 'std::__cxx11::string std::__cxx11::to_string(long double)':
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6509:50: error: 'vsnprintf' is not a member of 'std'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
c:\users\destari\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits\basic_string.h:6509:50: note: suggested alternative: 'isprint'
     return __gnu_cxx::__to_xstring<string>(&std::vsnprintf, __n,
                                                  ^~~~~~~~~
                                                  isprint

*** [.pio\build\adafruit_feather_esp32_v2\lib99a\LibPrintf\LibPrintf.cpp.o] Error 1
```

Destarianon, 2023-03-07:
My apologies for missing that. I moved an existing project to a different board and didn't even notice the platform difference.

phillipjohnston, 2023-03-07:
That's all right, I like the easy issues!